spacelab-ufsc/ttc2 #19: Firmware: Develop a driver of the INA226 sensor

mgm8, 2021-04-11:
[https://www.ti.com/product/INA226](https://www.ti.com/product/INA226)

mgm8, 2022-03-09:
We already have an initial structure of the driver: [https://github.com/spacelab-ufsc/ttc2/tree/dev_firmware/firmware/drivers/ina22x](https://github.com/spacelab-ufsc/ttc2/tree/dev_firmware/firmware/drivers/ina22x)

yurigazzoni, 2022-03-12:
I will need the values of Rshunt and of the maximum expected current to calibrate the register, to correctly measure the current, @mgm8.

mgm8, 2022-03-12:
> I will need the values of Rshunt and of the maximum expected current to calibrate the register, to correctly measure the current, @mgm8.

The Rshunt value is available in the schematics (sheets 3 and 4): [board_full_pdf.pdf](https://github.com/spacelab-ufsc/ttc2/blob/dev_hardware/hardware/outputs/board_schematics/board_full_pdf.pdf)

[image]

**Obs**.: All sensors use the same value for the shunt resistor (0,1 Ohm).

For the maximum expected current, we can consider 1000 mA for the radios, and 100 mA for the MCUs (but maybe we will have to calibrate these values when testing the firmware in a PCB).

**Tip**: Put these variables as parameters of the [configuration structure](https://github.com/spacelab-ufsc/ttc2/blob/72f6e7801ca760e01a35dbdb058a0b54ce0dcb9b/firmware/drivers/ina22x/ina22x.h#L77) of the driver. This way it will be easier to change the values in the higher layers of the firmware or reuse the driver.

miguelboing, 2022-06-28:
INA22x Battery Test:
* Use calibration values for uC only:
    config.cal= 10240;
    config.lsb_current = 5e-6;

- [ ] 1. Initialize the ina22x sensor (ina22x_init);
- [ ] 2. Configure with (ina22x_configuration):
    - [ ] avg_mode = 1 sample; bus_voltage_conv_time = 140u; shunt_voltage_conv_time = 140u; op_mode = Shunt and Bus, Triggered;
    Check if config_mask is correct;
    - [ ] avg_mode = 128 samples; bus_voltage_conv_time = 588u; shunt_voltage_conv_time = 588u; op_mode =Shunt Voltage, 
    Check if config_mask is correct;
- [ ] 3.  Calibrate ina22x sensor for uC (ina22x_calibration);
- [ ] 4. Read Configuration Register (ina22x_read_reg);
- [ ] 5. Read Calibration Register (ina22x_read_reg);
- [ ] 6. Read manufacturer ID (ina22x_get_manufacturer_id);
- [ ] 7. Read DIE ID (ina22x_get_die_id);
- [ ] 8. Read all measurement registers (ina22x_read_reg);
- [ ] 9. Read all raw measurement with functions and compare with item 8;
- [ ] 10. Read converted measurements and compare with item 9;

miguelboing, 2022-07-20:
[Tutorial](https://training.ti.com/programming-power-settings) used to program the calibration register.

commit f5a17728 (2022-05-17): firmware: drivers: ina22x: Minor fix #19
commit 732f154b (2022-08-01): firmware: drivers: ina22x: Adding changes from testbench, closes #19
commit 627ff9d5 (2022-08-02): firmware: drivers: ina22x: Minor fixes #19
closed 2022-11-04 by mgm8