spacelab-ufsc/ttc2 #22: Hardware: Review the "Design rules" of the altium project for DRC inspection

andrempmattos, 2021-04-15:
Check if they have reasonable values and if are similar to the OBDH's rules.

YanAzer, 2021-04-16:
@andrempmattos it may be usefull to import the OBDH's rules or even the original TTC's rules used in previous project.
https://www.altium.com/documentation/knowledge-base/altium-designer/import-or-export-design-rules

andrempmattos, 2021-04-24:
Since I opened the previous TTC project and started designing from that, the rules are the same. The issue is intended to remind us to check the rules before the final release and manufacturing.

YanAzer, 2021-05-05:
@andrempmattos i checked the DRC and it looks fine, since the TTC project was accomplished successfully with the same set of rules we are secure to move on with the final reviews for the first release.

closed 2021-06-04 by andrempmattos