spacelab-ufsc/ttc2 #38: Hardware: Review BOM and print the full schematic again (to include BOM in pdf)

andrempmattos, 2021-05-02:


YanAzer, 2021-05-12:
U4, U5, U9 and U10 missing partnumber parameter with the right format, the "Manufacturer_Part_Number" format is not acquired automatically by the BOM generation.

[image]

YanAzer, 2021-05-12:
- [CRCW08050000Z0EA](https://www.digikey.com/en/products/detail/vishay-dale/CRCW08050000Z0EA/1175125) with wrong description, change 1/16W to 1/8W.
- [RC0805FR-0747KL](https://www.digikey.com/en/products/detail/yageo/RC0805FR-0747KL/727973?s=N4IgTCBcDaIEoGEAMAOJBWAYnAtEg7ACz4DSAMiALoC%2BQA) with wrong description, change 1/16W to 1/8W.
-[SR732ATTER100F](https://br.mouser.com/ProductDetail/KOA-Speer/SR732ATTER100F?qs=%2Fha2pyFaduh9lG2DkCMz7sWyw6rZzaOWZ2KBl%2FK%2F3PYRVKaphDRAKw%3D%3D) with wrong description, change 1/8W to 1/3W.
-[53398-0671](53398-0671) with wrong partnumber should be 53261-0671 (right angle version)
- [AC0805FR-13100KL](https://www.digikey.com/en/products/detail/yageo/AC0805FR-13100KL/13694083?s=N4IgTCBcDaIIIGEAMAOJBWAYgJQLQEYBmfJJAaQBkQBdAXyA) with wrong footprint, change to 0603
-[BLM21PG300SN1D](https://www.digikey.com/en/products/detail/murata-electronics/BLM21PG300SN1D/584262?s=N4IgTCBcDaIEIBkCyYCMAFA4gZgAy4GUA5VAERAF0BfIA) with wrong description, change to 4A
- [SSW-126-04-G-D](https://www.samtec.com/products/ssw-126-04-g-d) don't forget to add quantity to 2

commit ab381d1d (2021-06-03): hardware: libraries: Updating ID and descriptions of components on the schematic library
hardware: sources: Updating schematics and layout with new component parameters
hardware: outputs: bom: Fixing BOM close #38
closed 2021-06-04 by andrempmattos