spacelab-ufsc/ttc2 #60: Firmware: Develop a driver for the SX127x transceiver

mgm8, 2021-12-14:
Reference: [https://www.semtech.com/products/wireless-rf/lora-core/sx1278](https://www.semtech.com/products/wireless-rf/lora-core/sx1278)

miguelboing, 2023-01-05:
- [ ]

miguelboing, 2023-01-05:
Implementation progress:

- [ ] sx127x_init(void);
- [ ] sx127x_rx_init(void);
- [ ] sx127x_tx_packet(uint8_t *data, uint16_t len);
- [ ] sx127x_rx_packet(uint8_t *data, uint16_t *len);
- [ ] sx127x_wait_irq(uint8_t irq_mask);
- [ ] sx127x_enter_standby_mode(void);
- [ ] sx127x_enter_sleep_mode(void);
- [ ] sx127x_clear_interrupt(void);
- [ ] sx127x_set_frequency(uint32_t freq);
- [ ] sx127x_set_rf_param(uint8_t bw, uint8_t cr, uint8_t sf, uint8_t crc);
- [ ] sx127x_set_preamble_len(uint16_t len);
- [ ] sx127x_set_header_mode(uint8_t mode);
- [ ] sx127x_set_payload_len(uint8_t len);
- [ ] sx127x_set_tx_power(uint8_t pwr);
- [ ] sx127x_set_rx_timeout(uint16_t symb_timeout);
- [ ] sx127x_read_rssi(uint8_t mode, uint8_t rssi_val);
- [ ] sx127x_power_on_reset(void);
- [ ] sx127x_config(void);
- [ ] sx127x_set_ant_switch(sx127x_mode_t mode);
- [ ] sx127x_set_fifo_addr_ptr(uint8_t adr);
- [ ] sx127x_enter_rx_mode(void);
- [ ] sx127x_enter_tx_mode(void);
- [ ] sx127x_write_fifo(uint8_t *data, uint8_t len);
- [ ] sx127x_read_fifo(uint8_t *data, uint8_t *len);
- [ ] sx127x_set_tx_interrupt(void);
- [ ] sx127x_set_rx_interrupt(void);
- [ ] sx127x_read_reg(uint8_t adr, uint8_t *val);
- [ ] sx127x_write_reg(uint8_t adr, uint8_t val);
- [ ] sx127x_burst_read(uint8_t adr, uint8_t *ptr, uint8_t len);
- [ ] sx127x_burst_write(uint8_t adr, uint8_t *ptr, uint8_t len);
- [ ] sx127x_spi_init(void);
- [ ] sx127x_spi_transfer(uint8_t *wd, uint8_t *rd, uint16_t len);
- [ ] sx127x_spi_write_byte(uint8_t byte);
- [ ] sx127x_spi_write(uint8_t *data, uint16_t len);
- [ ] sx127x_spi_read(uint8_t *data, uint16_t len);
- [ ] sx127x_gpio_init(void);
- [ ] sx127X_gpio_write_reset(bool state);
- [ ] sx127x_gpio_read_nirq(void);
- [ ] sx127x_delay_ms(uint32_t ms);

commit 76c6e2a0 (2021-12-20): firmware: drivers: sx127x: Implementation of the register access functions, and general fixes #60